Tympan/Tympan_Rev_F_Hardware #1: RTC Battery Life

biomurph, 2024-02-08:
I saw that the recommended coin for RTC is 2032 (20mm diameter).
We currently use 12mm diameter 1225. the difference is
2032 ~220mAh
1225 ~40mAh.

Then poking the form, I saw these posts
https://forum.pjrc.com/index.php?threads/teensy-4-x-rtc-battery-life.68148/
https://forum.pjrc.com/index.php?threads/beware-t4x-rtc-power-consumption.64216/#post-257979

tldr: Expected battery life in days on the Teensy 4.1

Days @30uA
CR1220 = 50 
CR2032 =  305
CR2450 =  833

That is not long, even for the 2032, given we have these in the field. Are we getting any feedback on this from users?

chipaudette, 2024-02-09:
I haven't heard any feedback, either way.

I would support using a higher capacity coin cell, but they are so much bigger that I bet it'll l be too hard to fit.

biomurph, 2024-02-12:
PJRC recommends a 2032 coin cell (20mm dia, 3.2mm thick).
Just can't fit that diameter on the board.
At least it's easy to replace?

chipaudette, 2024-02-13:
The army users are used to the Rev D, which presumably had a much longer RTC battery life.  Ive become more fearful that this short life with Rev E will cause them disappointment.

On the teensy forum thread about the high RTC power consumption, folks talked about switching back to an i2c RTC instead of using the built in rtc.  One individual talked about the "I2C RTC - RV-3028-C7".  Or, I think that there are a few qwiic RTC from sparkfun.

@biomurph, Have you used any of these before?

biomurph, 2024-02-13:
Yes, I have used RTC modules and also incorporated RTC chips in projects.
Here are some examples
https://www.adafruit.com/product/3028
https://www.adafruit.com/product/5188
SparkFun also has a bunch. Many with the Qwiic | Stemma connections.

I guess another route would be to use a rechargable coin cell, and incorporate it into the charging circuit that we already have?