Tympan/Tympan_Rev_F_Hardware #2: BLE Radio OTA DFU Only Works With Adafruit Bluefruit Connect App?

biomurph, 2024-03-25:
It has been the case that the only OTA DFU process that completed successfully has been done with the Adafruit Bluefruit Connect  App. This thread will record the results of testing OTA DFU using various methods. Each test begins with a radio module that is direct from production testing, flashed with the TympanRadio_baseFirmware.ino sketch.

OTA DFU variables include
Operating System
* iOS
* Android
File Base Used
* Arduino IDE Menu Select: Sketch > Export Compiled Binary
* Arduino IDE .zip file created during normal compilation process
App Used
* Adafruit Bluefruit Connect App
* nRF Toolbox App or nRF Connect App

## Android OS
Tested using Google Pixel 6
* Adafruit Bluefruit App uploads OTA successfully
* Nordic nRF Toolbox App
  * Fails using Arduino Export Compiled Binary after DFU Initialization, at the beginning of the upload process
  * Fails using Arduino zip created during compilation after DFU Initialization, at the beginning of the upload process
* Nordic nRF Connect App 
  * Fails using Arduino Export Compiled Binary after DFU Initialization, at the beginning of the upload process
  * Fails using Arduino zip created during compilation after DFU Initialization, at the beginning of the upload process

Successful OTA DFU is possible, but only via the Adafruit Bluefruit App.
I uploaded 3 times in a row with 3 times success.

chipaudette, 2024-03-25:
I still can't get Adafruit Bluefruit Connect to accept any of the file managers on my phone.  I have both the default google "Files", I have Google Drive, and I downloaded (per the Adafruit error message) Dropbox.  No joy.

I went into the settings for Bluefruit Connect and made sure that I was allowing access to "media and files".  Still, no success.

Specific error is "Unavailable.   File browser app not [sic] has not been found on your device.  You should install an app like Dropbox, GoogleDrive..."

Googling this error only reveals the GitHub repo for this App.  I had been hoping a Adafruit forum posting might show up in the Google results.  Nope.  :(

biomurph, 2024-03-26:
@chipaudette Are you able to test with a different phone?
I will test with my burner today. It is an LG phone, I think.

chipaudette, 2024-03-26:
I just tested with my ancient phone via the Adafruit app.  Unlike on my normal phone, it correctly found a File Browser App, so I was able to select the HEX and DAT files.  Unfortunately, it failed to get very far in the actual update process.   :(

This was for your nRF52840 stand-alone board, not a Rev F.  I'm a little hesitant to try this on a Rev F as we risk locking up its nRF unit until we get the nest working.  But, I suppose that I ought to give it a try anyway?  What do you think?

biomurph, 2024-03-26:
I have always been able to bring the nRF51 back to life with the bed of nails and the J-Link. That process erases the whole chip and starts from scratch.
You should get email notifications from USPS about the package shipping and arrival at Creare.

biomurph, 2024-03-27:
Unfortunately, I am unable to load Bluefruit Connect App on my old backup phone... 
Guess I need a new backup phone...

chipaudette, 2024-03-29:
The project had an older Android phone that I was able to borrow.  It runs the update portion of the Adafruit app just fine.  I've successfully updated the firmware a couple of times.  It ain't fast, but it does the job.   Excellent.