Tympan/Tympan_Rev_F_Hardware #3: Tympan Radio Production Firmware

biomurph, 2024-03-25:
There are many challenges that come with switching radio modules from the BC127 to the nRF52840.

One of the biggest ones is, what firmware to use?

1. We could build our own software that mimics the behavior of the BC127
2. We could use existing firmware that provides AT Command API
4. We could build our own firmware with AT Command protocol or our own custom API

Number 3 seems like a bad idea.
We are stuck between 1 and 2. 
1 means that we will be responsible for making sure the radio firmware functions and have to build the Command and Data coms
2 means that the Tympan Library will have to be modified to use the AT API instead of the Melody API from BC127 days.

Part of the question in the first place is, where can we get the AT Command Firmware to flash on our radios?
Turns out the Adafruit Bluefruit App lists firmware releases in its Firmware Update section. There are releases for

- BLEFRIEND
- BLEFRIEND32
- BLESPIFRIEND

@chipaudette, you had some success using an Adafruit BLE Friend in your initial testing of the nRF52840 radio functionality. We could potentially use the Adafruit supplied firmware to turn our radio into a BLE Friend of some kind?

I attempted to upload the latest version of BLESPIFRIED and BLEFRIEND, but they both failed to start after initializing the DFU bootloader. I don't think that the Adafruit BLEFRIEND is going to work out.

chipaudette, 2024-03-25:
Presumably the BLEFRIEND32 also failed?

Chip

biomurph, 2024-03-25:
I didn't try it, just assuming it would. Looking at the Adafruit products that deploy this firmware, I think it is a hardware target issue. The code just isn't built for out module? That's my guess.