Tympan/Tympan_Rev_F_Hardware #5: Typo on BOM - Power Switch Part Number

eyuan-creare, 2024-08-28:
[BOM ](Tympan_Rev_F/Production/Tympan_Rev_F_BOM.xlsx
)shows CL-SB-22A-11T, which is the Rev-E side switch, but should be CL-SB-22A-11T, the Rev-F top switch.

[image]

biomurph, 2024-08-28:
@eyuan-creare 
The slide switch with lever on top was a mistake.

biomurph, 2024-08-29:
We did have some customers who said that the side switch was too easy to flip when they are trying to adjust the volume knob, so this mistake might be actually better? 
Right now, in have inventory with the top slider. I was able to cut the case top prints that I made, but a redesign of the top case to make it easy to flip the switch is in order for the balance of my top case prints.