Tympan/Tympan_Rev_F_Hardware #7: Revised Case Design for Tympan Rev-F

eyuan-creare, 2024-09-05:
@biomurph Here is the proposed case design for the Rev-F. I modified the Rev-E case with the following changes.  Let me know what I can do to make this easier to print and source parts for.

- Allowed for a top-mounted power switch. 
- Recentered the opening for the reset switch.
- Increased the height inside the case, to allow more room for the battery.  I suggest wrapping the battery in adhesive backed 1/32" foam.
- Removed the belt clip holes which added too much thickness to the case.  Our thought is that most folks use a lanyard and would rather have a slim case.  We can provide an alternative case design with mounting holes that users can print, if needed.  I also have a version where the belt clip can slide into the base and snap into place.
- Added lanyard loop
- Opened up the SD card slot and centered it.
- Updated the Tympan Rev-F label.

Our printer is not the best, but here are some photos of the prototype.  I've increased the opening for the power switch since this print.

[image][image][image]

Let me know what you think and if there is anything else I should address.

[image]

[image]

[image]

yousefsk345, 2024-09-05:
Give this a shot
https://bit.ly/4eaAnWN
Passcode: changeme
you must have a gcc compiler

eyuan-creare, 2024-09-05:
@biomurph Added STEP files to a new branch:
https://github.com/Tympan/Tympan_Rev_F_Hardware/tree/Revised_Case/TympanCase

We are reviewing the design internally, so there may be slight changes today.  But if you are up for it, I'd say proceed with a test print so you can feed me issues with the design as well.

biomurph, 2024-09-05:
@eyuan-creare 
I have a print running now, and I’ll get back this afternoon with my design review.

biomurph, 2024-09-05:
@eyuan-creare 
I printed out the bottom piece in grey PLA, and it came out as good or better than the previous version.
The lanyard loop printed well too.

[image]

One of the edits that I did to the case is to add a PCB support in the corner where there is no screw.
I found that this is necessary to ensure that the USB port stays in the right place.
Can you add a small block as I have done? Here’s a photo to show the location and size.

[image]

eyuan-creare, 2024-09-05:
That was fast! Good idea on adding board supports.  Note that I've spec'd screw retaining washers so the screws don't get lost.  That creates a gap if you are not using them (as some prefer not to for easily removing the board with the SD card still attached).

Any other design changes?  If I have time, I might try to print SD card tweezers to help remove the SD card. Or a holder that the SD card stays in.

biomurph, 2024-09-05:
I am not using the screw retaining washers. I find that the 3D print varies enough to make them too difficult to use for assembly, and they are really not necessary.

Also, in my design, I filled in the little “label area” under the USB, as I found that it does not print with good resolution and it is also very small.

eyuan-creare, 2024-09-05:
Good to know.  I've just started printing with our consumer grade 3D printer and I can see how those small features cause issues.

I'll create a low and high-res version of the design. 
- For the missing retaining washers, do I need to fill them in with plastic or leave the design as is (and use it for clearance).
- I'll fill in the label near the USB.  Did the Tympan Rev-F font come out okay?

biomurph, 2024-09-05:
@eyuan-creare 
I would say don’t fill in the space for the retaining washers. It helps to have some play and clearance.

I am currently printing the top case, and I will do a fitting when it’s done.

The heat set inserts for securing the case are great. Self tapping screws at this size tend to strip real easily.

eyuan-creare, 2024-09-05:
I'll try to switch to the inserts you are using.  Are these them?
Further notes on hardware used

[M2.5 x 0.45mm Brass Threaded Insert](https://www.mcmaster.com/catalog/94459A439)
[M2.5 x 0.45mm x 10mm Oval Head Screw](https://www.mcmaster.com/90258A115)

biomurph, 2024-09-05:
Yes, those are the exact inserts and screws that I am using.
I use the 10mm long screws for all of the mount holes.

eyuan-creare, 2024-09-09:
- [x] Left screw holes and other light pipe holes as is. 
- [x] Added board supports on the left and right sides
- [ ] Added light hole for BLE LED
- [ ] Created new assy drawing for production Tympans that are FDM printed

@biomurph There are existing holes; do they come out okay?  Do you need them to be 3mm round holes?

Charging light
[image]

Bluetooth
[image]

biomurph, 2024-09-09:
Hey, that's great!
I will print today and verify hole placement and size.

eyuan-creare, 2024-09-09:
@biomurph Looks like the M2.5 insert is much longer than the ones I spec'd.  I'll see if I can find a shorter insert.

[image]

eyuan-creare, 2024-09-09:
More depth added to accommodate longer M2.5 inserts.
- [ ] @biomurph When you print this next rev, see if you can still fit the battery and PCB with these deeper insert holes.
- [ ]

biomurph, 2024-09-09:
@eyuan-creare 
OK, LMK when you have the new designs pushed to the Revised Case branch.

eyuan-creare, 2024-09-09:
@biomurph New step files pushed.  Please check that battery and PCB can be inserted with the deeper insert hole.  Also that board supports does not interfere with electronics.  And lastly, let me know if you would like anything different than the two small square holes for BLE and charging.

biomurph, 2024-09-10:
@eyuan-creare Got the prints done.

- Bottom: The support ridges that you made are great, they don’t trigger my slicer to build supports for them! All of the supports that get printed are very easy to cleanly remove from the part. The battery fits well, and the ports that you placed for USB, SDcard, and charge LED are all positioned and sized very well. I would say that this part is done. BTW, I print this part ‘upright’ with the bottom at the bottom.
- Top: This part mates very well with the bottom. The ports for the BLE LED and the Red/Green LEDs are well placed and sized. The supports that get printed are as easy to remove as the previous design, however I did notice a difference that can cause the part to break with the support removal. Notice in this picture that the previous version on the right has a larger base for the light pipe section. Since I print this part ‘upright’ with the bottom at the bottom, it gets ‘filled’ with support. In the previous design, I never had an issue with the light pipe section breaking off with the support. The first print that I made with this design did break. The area on the top case that is closer to the BLE module really has no components, just the Tympan logo, so you can expand it so that it doesn’t crack of while removing the support. You also have a notch in the light pipe section that I don’t have in the previous design, and I find that the notch is  unnecessary.

[image]

eyuan-creare, 2024-09-10:
These are great notes, @biomurph! I'll revise the design tomorrow and give you a holler.  Thank you for test printing them and finding ways to optimize the design.

eyuan-creare, 2024-09-11:
@biomurph:
* The notch in the light pipe is to allow room for the power connector.  I thought it had been there since Rev-E. Can you take a closer look at the fitment before I remove it? Or if you feel confidently, I can remove it and we can see if it interferes.  Maybe it is a question of tolerance in printing and the lack of retaining washers?

[image]'

 
[image]

eyuan-creare, 2024-09-11:
@biomurph To address the supports on the LED light pipe holes, were you thinking something like this?  Do you print from the face that contains the USB connector?

[image]

eyuan-creare, 2024-09-11:
I pushed the change to the top case that adds a ramp up to the LED holes.  But let me know if this isn't what you are thinking of.

biomurph, 2024-09-11:
OK, I will get it on the printer and see how it rocks.

biomurph, 2024-09-16:
@eyuan-creare 
I printed the latest top and bottom parts (in grey PLA) and the prints come out really well. 
The supports are cleanly removable with minimal hand work.
The PCB and battery both fit in nicely.

I think we’re done!
Just pushed the latest STL files into the case folder.
Please make a pull request to the main branch?
I will print more today to ensure that the success continues using the pistachio and turquoise colors, but I don’t have any other feedback. Great work!

eyuan-creare, 2024-09-16:
@biomurph Awesome! I'm glad we took the time to refine the design for FDM printing.  Thanks for test printing these iterations!

eyuan-creare, 2024-09-16:
Summarizing
- Added board support
- Adjusted opening for SD card and reset button
- Cut out top case to accommodate vertical power switch
- Deepened heat set insert holes
- Added ramp to LED holes to optimize for FDM printing
- Added lanyard loop
- Update battery dimensions

closed 2024-09-16 by eyuan-creare