Tympan/Tympan_Rev_F_Hardware #9: Review enclosure design

eyuan-creare, 2024-09-24:
Noticed a couple of issues with the new enclosure (recvd Sept 24, 2024).
- Heat sink insert needs thicker boss
- Square light pipe hard to press in (but goes in well after initial push)

eyuan-creare, 2024-09-24:
Charging LED: BIVAR PLPQ2-188-FF
- Requires panel cutout of 0.080in sq with 0.008in radius.
[image]

This matches the existing design:
[image]

Leave as-is for now and see if it is an issue with future prints.
-

biomurph, 2024-09-24:
@eyuan-creare 
I have not been using a square light pipe in that position. 
The charge LED shines through quite well as it is.

You mention the heat-sink insert, the blue in your rendering is on the screw side of the join, not the thread side?

eyuan-creare, 2024-09-24:
Yes, my mistake.  The clearance for the screwdriver has a thin wall that blew through.  I've added thickness there.  I will correct the typo in the issue.  Light pipe won't be present for typical production units, correct.

eyuan-creare, 2024-09-24:
Uploaded new STL's (units: mm), STEP, and PDF of assembly drawing.  Pushed new files to branch and merged with trunk.

closed 2024-09-24 by eyuan-creare