Tympan/Tympan_Rev_F_Hardware #11: Rev_G Wishlist

chipaudette, 2024-09-26:
We currently have no specific plans for making the next revision of Tympan.  But, if we were to make another revision someday, what hardware changes would we like to have?

@biomurph, @eyuan-creare, @cab-creare-com, etc...please add a post here to this issue and, once I understand the request, I will then collect them into the list below.

Desired changes for Rev_G:

* Add breakout pads/pins for the JTAG interface.  This will allow us to use a debugger while developing firmware.
* (others?)

b-graybill, 2024-09-26:
Thanks, Chip!

cab-creare-com, 2024-09-26:
Yes to Bluetooth Audio

* Also requested by potential users: WiFi. (Particularly to support web-based speech recognition APIs.)

eyuan-creare, 2024-09-26:
- Push-Push SD card slot
- High Density breakout connector for prototyping that exposes more uC pins
- (2) AICs (Seems that we often need 4-channel recording
- USB-C with Fast Charging
- I2C QUIIC connector

**Nice to Have**
- Power Management IC that has a shipping mode that activates when charger is plugged in
- 8-channel I2S [see Chip's comment on issue](https://github.com/Tympan/Tympan_Rev_E_Hardware/issues/1#issuecomment-618677266)].
- Pads for optionally adding battery clips for a single 18650 lithum cell
- Pads to expose more GPIO pins
- Develop driver for fuel gauge.
- Snap together case instead of screws
- Ship with demo firmware that cycles through (n) applications (Basic Gain, Pitch Shifting, 8-band equalizer, ....)

cab-creare-com, 2024-09-26:
* Power management fix to allow programming when powered via USB without battery (or when battery is discharged).

chipaudette, 2024-09-26:
@eyuan-creare , Rev F already as the quiic connector!

eyuan-creare, 2024-09-26:
> @eyuan-creare The quiic does not have a port in the current case design. Might be a good idea to post a design mod with a quiic port for folks to print if they want to?

Good idea.  Are the pads for a right-angle QUIIC at the edge of the board?

cab-creare-com, 2024-09-26:
Before we get too far with this, the correct name is Qwiic:  "https://www.sparkfun.com/qwiic".

eyuan-creare, 2024-09-26:
I knew you or Chip would correct me, so this was the most efficient path to the correct spelling ;)

b-graybill, 2024-09-26:
Large flat section on case for easy labeling with large font lettering

biomurph, 2024-09-26:
Here is the Qwiic connector position
[image]

cab-creare-com, 2024-10-02:
Improve RTC coin cell battery life. (Re: https://github.com/Tympan/Tympan_Rev_F_Hardware/issues/12#issuecomment-2388928978)

eyuan-creare, 2024-10-02:
> Improve RTC coin cell battery life. (Re: [#12 (comment)](https://github.com/Tympan/Tympan_Rev_F_Hardware/issues/12#issuecomment-2388928978))

Maybe a rechargeable coin cell?  Shorter run time, but rechargeable.  Hopefully find one with UN38.3 certs.

eyuan-creare, 2024-10-02:
Protective Circuit for Coin Cell [minor]
- Add resistor inline to prevent overcharging the coin cell when the Schottsky diode fails.  Referenced in many designs and datasheets for suggested battery backup circuits.

[image]

chipaudette, 2024-10-02:
We may have had someone damage their Tympan by applying 5V to the audio input jack (the pink jack).  Perhaps we should add over-voltage on this connection?

eyuan-creare, 2024-10-04:
- Brown out protection.  USB Serial may not initialize if battery is low (how low?)

eyuan-creare, 2024-10-15:
(2 to 3) Physical push buttons on the board

chipaudette, 2024-10-17:
> * Upload hex files of each example sketch so that users can quickly try them out.  Right now, takes several minutes to upload a new sketch.

@eyuan-creare , I know that you've always been a bigger fan of the HEX files than I have been.  In my experience, unless you hold their hand through the process, uploading a raw HEX is just too abstract for inexperienced users.  My biggest issues are:

* Just finding Teensy.exe requires digging down into the bowels of the Arduino's installation directory structure...which makes many users very uneasy.  As an example, below is the path to it on my computer.  It's deep!  And, more troubling, "AppData" defaults to being hidden by Windows...so, yikes: 

 `C:\Users\Chip\AppData\Local\Arduino15\packages\teensy\tools\teensy-tools\1.59.0`

* Keeping the HEX files up-to-date is always difficult.  Currently, it would have to be done manually (with every release?), which seems like an impossible task given the sheer number of our examples.  Maybe we could set up an automated process, but maintaining that process isn't easy either (KXM and I tried this years ago...the system didn't last but a few months).

What's your experience here?  I know that you've had more positive experiences.  Tell me more!

eyuan-creare, 2024-10-20:
> @eyuan-creare , I know that you're a fan of HEX files. More of a fan than me, for sure. In my experience, the process is never as easy as one hopes, at least from a new-user perspective.

Good point.  I think there are two things that I am considering:
- Hex files document tested code, independent of differences in the compiler or libraries.  When doing research, it's good to know that when a hex file is uploaded, it is the exact same version as when it was last used.
- For users who may be using the Tympan as an open source research tool, rather than a development platform, there may be an easy way to streamline the hex upload process, using a bat file or tablet app to kick off the CML version of the Teensy.  Perhaps to some users Arduino's IDE may be daunting.

In terms of uploading hex files, we could start by including a hex whenever we release and test a new software feature (not necessarily keep all the hexes up to date with the latest rev).  Just a snapshot in time when things had worked.  Then when new hardware is released, upload them when we happen to test and verify a script is stable, for those sketches we commonly use.  Perhaps not the time yet to put this into play.

chipaudette, 2024-10-20:
@eyuan-creare , you e painted a good picture, especially in drawing a difference between researchers vs developers. Continuing with this idea, it might suggest keeping the collection of hexs in their own repo.  This repo being aimed at users rather than developers.  The docs for the repo (readme, for example) could be aimed at this kind of user..."how to use these files to easily discover the Tympan".  I like that.  A clear purpose, with words that help people realize that this is what they want.

Additionally, using a separate repo for the hexs addresses my concern of the hexs being hard to maintain, hard to keep them matching the latest code base.  Being a separate repo, it isn't necessarily expected that the Researcher repo (ie, hexs) is always perfectly aligned with the Developer repo (ie, Tympan_Library).  The hex repo could have its own releases, with notes as to what standard the hexs were made to.  Having a hex-only repo gives up more options for keeping hexs for the diff versions of Tympan, too, if we desired.

What do you think about the hexs being in their own "For Researchers" repo?  From your viewpoint, does this achieve your vision better or worse?

biomurph, 2024-10-21:
@chipaudette 
I like the idea of a separate repo. That will make it easier to maintain and build.
I also get you when you say the Teensy loader can be a tedious tool to work with.
Teensy Loader is available as a stand alone app, it seems.
https://www.pjrc.com/teensy/loader.html
has anyone tried it with Teensy 4.1? If it works, we could incorporate it into a tutorial or guide to make the process more straightforward?

cab-creare-com, 2024-10-21:
BAyotte has been loading Hex files directly on a Tympan Rev F, following the instructions in "https://github.com/Tympan/Docs/wiki/Program-the-Tympan-from-a-Single-Hex-File#uploading-a-hex-file-to-the-tympan-1".  That involves digging into the Arduino app data directory that @chipaudette mentioned. 

I'd be in favor of adding instructions for how to use the standalone Teensy loader.

chipaudette, 2024-10-21:
Oh man!  One can download the Teensy Loader directly from the web!  That totally addresses my concern!  Thanks @biomurph!

biomurph, 2024-10-21:
There is an active tutorial on pjrc that uses the stand alone be loader to flash micro python... It would need some basic testing.

cab-creare-com, 2024-11-13:
The battery charger/Tympan power circuit needs some attention. **Charging the battery should not take priority over powering the Tympan main functionality.**

From @chipaudette:

> In looking at the battery charger IC that’s built into the Tympan, I found some interesting limitations that might related to your recent experience with an overly-discharged battery.
> 
> * **Normal Operation:** The battery charger IC is designed to limit the current being delivered to the battery.  The limit is 500 mA.  This also means that the IC limits the current (from USB) being delivered to the Tympan itself.  **The Tympan will only be allowed to pull 500 mA from USB.**  We have always assumed that it is this limit that prevents our system from being reprogrammed if no battery is attached; we have assume that 500 mA must be not quite enough to push through all phases of reprogramming.
>
> * **Overly-Discharged Battery:**  Reading the datasheet for our charging IC, I see that it has a “pre-conditioning” charging phase, where it limits the current delivered to the battery to prevent damage to an overly-discharged battery.  For our charger, **if the battery voltage is less than 2.77V, the current from USB is limited to only 50 mA!**  Again, the limiting current to the battery is (in our system) the same as limiting current to the Tympan itself.  So, if the battery is drained below 2.77V, the Tympan will be limited to 50 mA.  **The system will not be able to boot on so little current.**

eyuan-creare, 2024-11-14:
For the meantime, how about a software version of UVLO?  When Tympan's analog battery voltage measurement is low, shut itself off.

biomurph, 2024-11-14:
@chipaudette Yes, we can use that topology to ensure the VBUS will power the system even when charging the battery.

cab-creare-com, 2024-11-14:
It's worth reviewing recommended circuits for USB & rechargeable battery powered devices. The Adafruit nRF52840 Express design is not ideal:
* When the battery is being charged and the MCU is powered on, the device potentially exceeds the allowable power draw from a USB 2.0 port (500 mA).
* When the MCU is powered off, the battery may not be charging as fast as possible. (Depends on R4 value and the battery specs.)

biomurph, 2024-12-04:
The Rev F was designed with a ‘gas gauge’ chip that is a better way to measure battery level than a resistor divider to analog input. However the one I chose went EOL and became unavailable at build time.
It’s worth it to find one that is supported by supplier and Arduino library. SparkFruit is good resource to find candidates.

eyuan-creare, 2025-07-02:
### Spreadsheet of Options
Please review and consider priority, effort, risk, and benefit.  Thanks!

https://crearellc.sharepoint.com/:x:/s/CreareExternalCollaborationSite/EdRqutMJEUNBpBE2oCk4FukBROEG6KA2YNnkSxv5aC9mIA?e=JckL8C