Tympan/Tympan_Rev_F_Hardware #12: USB Serial Port not shown

eyuan-creare, 2024-10-01:
There is no serial USB port shown in the Arduino IDE.  This has occurred on hardware S/N TYM-F003(@eyuan-creare) and F004 (@cab-creare-com).  

Regarding TYM-F003
-  Switching computers did not work. 
- Uploading a simple blink sketch (via compile and upload in the IDE, or via teensy loader directly).
- Bluetooth light would turn off, then turn on again, as if it was power cycling.
- Occurred after being fully charged.
- Was eventually resolved (much later) after pressing the program button (even though no serial port was shown).

Regarding TYM-F004
- Prior to the issue, was working correctly for several weeks.

eyuan-creare, 2024-10-01:
@cab-creare-com I'll try out your Tympan tomorrow and can give you a new one if you need it. Can you document the circumstances, since this is the second time this has occurred?  Arduino IDE version, sketch uploaded last, Tympan library branch, battery charged (i imagine you leave it connected).  Did you try pressing the program button? etc...

@chipaudette Anything to add about what you experienced with the Tympan that was giving me issues?

Might be worth comparing the Rev-E and -F PCB to see if there is anything about the Teensy boot process that is different.  I was surprised my Tympan TYM-F003 started working again after stubbornly resisting everything I tried.

chipaudette, 2024-10-01:
I've been suspicious of the coin cell battery as a potential source of bizarre system behavior.  What happens when the coin cell gets really dead?  Will the system work OK?  Would it be better to have no coin cell at all?

In the unit that I tried debugging for @eyuan-creare, I pulled the coin cell and did not replace it.  The Tympan started working shortly after that (also after a long press of the reset button).  Maybe removing the coin cell helped? Maybe it was irrelevant? 

So, if you have this problem again, can you document (1) whether a coin cell is present at all, and (2) what the coin cell voltage is (measured with a multimeter or whatever)?

biomurph, 2024-10-02:
@eyuan-creare 
Great work figuring that out! I hope that continues to be the root cause.

@chipaudette 
I am shipping Tympans without the coin battery installed. 
I do provide a coin cell, separate in the kit.
We had a discussion about how the Teensy 4.1 really drags the battery down fast, and so since we don’t need it, I include it as an option. The coin cell + is connected to the V-Bat pin on Teensy. It sounds like the internal RTC date and time is set according to the computer time on upload, and an optional battery on V-Bat will let Teensy keep time when powered down. This is from PJRC site:

> VBAT
> A 3 volt coin cell may be connected to VBAT & GND to allow the RTC to keep track of date / time while power is off. A CR2032 type battery is recommended, though other 3V coin cells may also be used.

chipaudette, 2024-10-02:
@eyuan-creare, nice sleuthing!

@biomurph, thanks for the clarification on shipping without coin cell installed.  I agree with your stated procedure when delivering Tympans.  So far, it's only the dosimeter folks who have expressed any need for the RTC...so your plan for regular Tympans is a good one, IMO.

eyuan-creare, 2024-10-03:
### TYM-F003 Issue back, but fixed it
Symptom: Only the LED for the BLE is lit, and it is slowly fading in and out.  No USB serial port listed in Arduino IDE
- Pressed the program button - no update on Teensy Loader
- Removed the coin cell, then pressed the program button - no update on Teensy Loader
- Replaced the coin cell with a brand new one. Pressed program button - no update on Teensy Loader
- Removed the battery. BLE LED started flashing.  Pressed program button but no update.  No USB serial port.
- 
THEN, swapped batteries.  No serial port listed.  Then I pressed the program button and a sketch was uploaded.  Serial port is now listed in Arduino IDE.
- Measured old battery at 2.5V.  Yikes!!! Looks like we need brownout protection.

eyuan-creare, 2024-10-03:
Resolution: Swapping to a battery with more charge allowed firmware to be "manually" uploaded with the program button, even though the serial port was not listed.

closed 2024-10-03 by eyuan-creare

eyuan-creare, 2025-03-25:
Reminder to develop new enclosure for Tympan with integrated Teensy.