Tympan/Tympan_Rev_F_Hardware #13: Making OTA DFU easy for users

biomurph, 2024-10-07:
When there are useful updates to the nRF53840 firmware, users need a simple way to grab the latest version so they can update their Tympan radios in the field. 
Asking them to download this entire repo and then dig through the files to find the right one is a bit onerous.

Github releases allow you to target a particular branch or commit for publication.
I propose to make a branch that will only contain a single folder that has the latest HEX and DAT files.
This seems to be the easiest way to keep everything in the same repo and easily accessible.

@eyuan-creare @chipaudette
Does anyone see an issue with this scheme?

chipaudette, 2024-10-10:
I'm open to it.  Personally, I was going to wait until we actually had a new version of the firmware that people needed.  I don't think that we've hit that threshold yet?

True, there has been one update to our nRF firmware, but that's only needed for people who will use the tablet software from the current contract (our "OpenHearing" project).  For you (non-Creare) sales to date, my guess is that none of those people who will ever use that OpenHearing tablet stuff?  So, there's no demand for this updating capability yet?

If you want to get ahead of the curve, your idea sounds good.  Or, if you want to wait until we have a more apparent need, that's cool with me, too.

biomurph, 2024-10-11:
Alright, I think I did it right.
@chipaudette @eyuan-creare 
If someone could just give the Read Me a once over to make sure it makes sense?
If it makes sense, then this issue can be closered.

chipaudette, 2024-10-12:
@biomurph, Thanks for doing this!

My only two comments are:

* It's unfortunate that the Readme says to go "HERE" but links to GitHub's normal release page, which only provides links to a ZIP or to a TAR.GZ archive.  My android device is always seems to be temperamental with handling ZIP (I haven't tried a TAR.GZ).  It's difficulty with ZIP always annoys me.  From this perspective, it's unfortunate the GitHub release page can't just have the HEX and DAT files.  Do you know of any way to get the release page to simply show the two files for download?

* All of your screenshots are fantastic to include.  Unfortunately, in my PC browser, they're huge!  They take up most of a page.  They're way larger than is comfortable (or needed).  I know that one can hardcode a smaller size (in pixels) to make the display more comfortable.  But, hardcoding image sizes might make the mobile view terrible?  Do you know anything about better managing image sizes in GitHub markdown?

biomurph, 2024-10-14:
@chipaudette 
Thanks for the feedback!
I will fix the link so that it just downloads the zip. That seems easiest.

I’ll see what I can do about the image size. I agree, they are too big!

biomurph, 2024-10-14:
@chipaudette 
Just updated. Turns out you have to use a little html to lock the image size to a fixed value.

Please check it out, and close this issue if there are no more issue? Thanks!

closed 2024-11-14 by biomurph