Tympan/Tympan_Rev_F_Hardware #14: nRF52 Firmware: why do we need to set a new MAC address in order to set new services?

chipaudette, 2025-02-16:
With the candidate v0.4.0 firmware, we can now add new BLE services.  Well, we can add the handful of services that I predefined.  It's a starting point!

The issue is that, for some reason, a Tympan user needs to first set a new MAC address before they can activate the new services.  I don't know why.   But, as shown in the Tympan_Library examples, it only works with the `setMacAddress` present:

https://github.com/Tympan/Tympan_Library/blob/ef2a8ca38ee0932c8ea00b2b12a70feb98240fdb/examples/02-Utility/Bluetooth/Tympan_BLE_BatteryService/Tympan_BLE_BatteryService.ino#L51

[image]

Why?  This shouldn't be necessary!  The MAC address should be completely unrelated to the new services.   Is it something in the AT_command processor code here in the nRF52840 code?

The goal is to find out why it only works if we include the `setMacAddress` line.  Then, we should fix this bug.

biomurph, 2025-02-20:
I have made extensive use of the [Adafruit nRF52840 core](https://github.com/adafruit/Adafruit_nRF52_Arduino) and I have not had to manually define a MAC address. I will look at the core to see if they are putting it in somewhere.

chipaudette, 2025-02-20:
Hey Joel, this was a bug report to myself.  I'm sure that I caused it myself with my other changes in v0.4.0.

Didn't mean to spin you up.  You can leave it for me to fight.

biomurph, 2025-02-20:
OK, I am flashing some Tympan radios today, and I will use the [latest release](https://github.com/Tympan/Tympan_Rev_F_Hardware/releases/tag/v0.3.1).

biomurph, 2025-02-20:
Or should we use the 0.4.0?
If so, I will make a new release.

chipaudette, 2025-02-20:
I like the new features in 0.4.0 a lot, but it doesn't seem right to ship with this known bug.

The bug doesn't show itself in any of the example baseline uses of the BLE system in the Tympan examples, but it does show up if you start to use some of the new features.  The mitigation of the bug is easy (always change the MAC first) but it's still a bug.  I'm not going to be able to do anything about it this week.

Because the bug still exists, that's why I haven't released 0.4.0.  So, I guess that you should stick with 0.3.1.